adafruit/Adafruit_Python_GPIO #16: Put package on PyPi

swstack, 2015-01-04:
I think it would be really useful to put this (and all of your) packages onto PyPi so people can actually `pip install` the packages directly instead of running setup.py manually.  You can do so by using the `sdist` command on setup.py.

Example:

```python setup.py sdist upload -r pypi```

With example .pypirc:

```
[distutils]
index-servers =
    pypi

[pypi]
repository: <repository-url>
username: <username>
password: <password>
```

tdicola, 2015-05-07:
Thanks for the suggestion and apologies it took a little while for me to review this one.  Great suggestion, and I just recently went through the process of getting setup on PyPi.  I'll look at getting this and the other python libs on PyPi in the near future.  One thing we need is an easy way to keep PyPi in sync with the versions we push on github, but for now we can just manually push versions as they're updated.  Thanks again for suggesting the change!

swstack, 2015-05-07:
Cool!  And yes, you hit on one of the main agonies of hosting your package on PyPi, keeping it up to date.  My company only has a couple on PyPi so we are still going through the manual process of pushing new code to GitHub, tagging master as our "release" and then uploading it to PyPi.  Maybe you could setup a branch that only gets merged to when a new release is ready and set up a post-commit hook to automate some of that process?  Just an idea.

Anyway, thanks for all of the cool libraries!

tdicola, 2018-02-27:
Closing this up as this has been on pypi for a while now under the adafruit-gpio name.  I still recommend installing from source though as it's a lot more clear when something fails to compile correctly (like any board-specific C modules).

closed 2018-02-27 by tdicola