adafruit/Adafruit_Python_GPIO #20: Simplify running tests

asomov, 2015-03-24:
Make it possible to run all the tests with
python setup.py test

tdicola, 2015-03-25:
Ah great point, for now I run things with nose or python's unit test discovery (`python -m unittest discover`).  When I get a chance, or if someone wants to take a stab at it, the docs for setuptools need to be investigated to see what's required to have the test command run everything.  I think it's probably as easy as adding a new option to the setup.py that points at the test directory.

tdicola, 2015-03-25:
Fixed with the pull request here: https://github.com/adafruit/Adafruit_Python_GPIO/pull/21  I also updated the readme to point at a new wiki page with info on running the tests.  Thanks again for contributing!

closed 2015-03-25 by tdicola