adafruit/Adafruit_Python_GPIO #27: FT232H code should use a lock to serialize access to the chip

tdicola, 2015-09-01:
Right now if you use FT232H code in a multithreaded app you need to be careful to serialize access to the library since two calls to the same FT232H at the same time will trample over each other reading data from the chip.  Consider adding a lock to serialize access at the library level.

mecworks, 2017-10-24:
I have the same problem using the MCP230xx module on a Raspberry Pi.

We need thread safe access to the underlying hardware.

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada