adafruit/Adafruit_Python_GPIO #32: I2C library assumes 8-bit registers

shamlian, 2015-10-26:
Some sensors (like the VL6180x from ST) and larger EEPROMs use 16 bits for their register addresses; the library assumes 8 bits. It would be great if class I2CDevice had a parameter in __init__ for the width of a device's register.

ChrisGermany, 2017-11-17:
Hi I am running in the same issue, did you solve the problem?

shamlian, 2017-11-18:
I did solve the problem, but not in this library. I opted to use an FT232H with the sensor I was controlling and did the transactions manually with a PC (instead of using the native interface on the Pi). Sorry.

ChrisGermany, 2017-11-20:
Did you ever try to change the FT232H.py to talk to 16-bit registers and ran into problems or you just decided to switch? I am using python on a raspberry pi, so I am kind of stuck with this library.

shamlian, 2017-11-21:
I was working on a quick and dirty demo; I had originally assumed that using a Pi with this library would be the shortest path to get it working, but when I ran into the problem, I added an issue to the tracker and then wrote something using an FT232H on my PC (without Adafruit's libraries). I have not spent time to fix Adafruit's library, since I haven't had the need to. Sorry and best of luck.

tdicola, 2018-02-27:
Yeah the I2C wrapper is using the low-level Linux kernel smbus and I2C interface which has an assumption of 8-bit register widths.  To work with less common larger registers you'll likely need to talk directly to the I2C kernel device (the /dev/i2c-0, /dev/i2c-1. etc. character devices) and send raw I2C data to it using read, write and ioctrl calls.  This is the reference if you're curious: https://www.kernel.org/doc/Documentation/i2c/dev-interface and the PureIO module is an example of making these ioctrl and other calls from python code:  https://github.com/adafruit/Adafruit_Python_PureIO  As an alternative though you might also look at more recent GPIO libraries that use the latest kernel's userspace I/O access facilities: https://github.com/sgjava/userspaceio

closed 2018-02-27 by tdicola