adafruit/Adafruit_Python_GPIO #36: I2C readList() function error in device address and missing restart condition

lifetyper, 2016-01-05:
I recently used the I2C readList() function to make a sequential read on AT24C04 EEPROM and got an Non ACK Error.Then I look into the code and I found something wrong in this funciton.

in Line 720 of FT232H.py：
## self._i2c_write_bytes([self._address_byte(True), register])

## Obviously,it's IMPOSSIBLE to WRITE a internal address(register) to a READ ADDRESS.

I think this function should work just like I2C readU8() funtion except changing the length.

Here below is my modification which works for me:

    def readList(self, register, length):
        if length <= 0:
            raise ValueError("Length must be at least 1 byte.")
        self._idle()
        self._transaction_start()
        self._i2c_start()
        #Write to a write addr here
        self._i2c_write_bytes([self._address_byte(False), register])
        self._i2c_stop()
        self._i2c_idle()
        self._i2c_start()
        #And generate a read restart here
        self._i2c_write_bytes([self._address_byte(True)])
        self._i2c_read_bytes(length)
        self._i2c_stop()
        response = self._transaction_end()
        self._verify_acks(response[:-length])
        return response[-length:]

bobbus77, 2016-01-12:
I had the same issue trying to use readList with a sensor.  I can confirm that your edits solved my problem.

-
On a slightly related note, I noticed there is likely another issue related to bulk reads in the tests.  Lines 70 & 71 of test_I2C.py read:
```
    def read_i2c_block_data(self, address, length):
        return [self._read_register(address+i) for i in range(length)] 
```
It appears as though we're incrementing the address instead of the register, meaning line 71 should be the following instead:
```
        return [self._read_register(address, register+i) for i in range(length)]
```
I noticed when I ran the tests that writeList is not being tested, which would explain how this bug got through.

FunkyFlier, 2016-04-12:
I was able to fix another issue with the readList function by commenting out the idle command in between the start and stop. The readList function worked for some devices. The MS5607 would not work with the idle condition in between the stop and the start. The devices which worked previously continued to work after removing the idle call. Here is the code:

```
def readList(self, register, length):
	if length <= 0:
		raise ValueError("Length must be at least 1 byte.")
	self._idle()
	self._transaction_start()
	self._i2c_start()
	self._i2c_write_bytes([self._address_byte(False), register])
	self._i2c_stop()
	#self._i2c_idle()
	self._i2c_start()
	self._i2c_write_bytes([self._address_byte(True)])
	self._i2c_read_bytes(length)
	self._i2c_stop()
	response = self._transaction_end()
	self._verify_acks(response[:-length])
	return response[-length:]
```

FunkyFlier, 2016-04-12:
To add on to what lifetyper said the idle call is also not needed to correctly generate a repeated start condition.

charlesvds, 2018-08-01:
I wish I'd found this earlier. I've just spent a long time finding this bug and came up with the exact same solution which works.
I wonder why the library hasn't been fixed yet ?
I'm going to give it a try.

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada