adafruit/Adafruit_Python_GPIO #40: Runtime Errors when reading large amounts of data

damitha1984, 2016-03-30:
Hi,

I'm using the Adafruit_GPIO library to talk to an SD card via SPI.

I have noticed that if I wanted to read back a large amount of data >1000 bytes by using either the .read or .transfer functions I get an error from the _poll_read functions.

"RuntimeError('ftdi_read_data failed with error code {0}.'.format(ret))
RuntimeError: ftdi_read_data failed with error code -9."

Any ideas why? 
When using a slow baud rate it works so do i need a delay before calling the _poll_read functions, i'll try this.   

Thanks

damitha1984, 2016-04-05:
The FTDI chip kept on locking up when I read large amounts of data, so I reverted to the FTDI D2XX driver and used the libMPSSE-SPI library.

I got the same results, until I change the Latency Timer value in Channel config to 2ms (>10 it didnt work)
FT_STATUS SPI_InitChannel (FT_HANDLE handle, ChannelConfig *config)

Now is there a way to do this from FT232H.py, then I can continue development in python.

Cheers

mpratt14, 2019-04-03:
@damitha1984 I know this is a really old issue, but I did a lot of work on the SPI methods recently, you may want to reinstall the library and try again.

It is worth noting that the FT232H and similar FTDI chips using their MPSSE engine are limited to 64 KB per command.
See #94 

If it's not that it might be:
 - a problem with libFTDI
 - hardware limit of your SD card (which is expected to run at lower baud in SPI mode)

damitha1984, 2019-04-04:
Hi mate, yes thats exactly what i did, reinstall the library and it worked

mpratt14, 2019-04-04:
@damitha1984 So you're using latest libFTDI with this library now?

damitha1984, 2019-04-04:
Well this was about 2 years ago. But I remember updating the driver and it
worked.

mpratt14, 2019-04-04:
@damitha1984 Yeah I see that, but the goal is to get it to work with libFTDI and the libusbK driver with this library, otherwise, we wouldn't consider this issue "fixed"

Some of the changes made to this library might solve your problem just as well. It also depends on how the functions are used....

For example, I added a function .bulkread, that might be better for your use case

And if not, it would be nice to know more in detail about how you were using this library and exactly what you were trying to do...maybe post a few lines of your script?

damitha1984, 2019-04-04:
Ok it was an old project. I'll have to try and find the source code. If I
find it I'll let you know.

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada