adafruit/Adafruit_Python_GPIO #41: Error message when quitting program with ft232h

Ivoah, 2016-04-28:
Why are these lines using check_call? https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/FT232H.py#L79-L80
It seems like just call would be better here, so that it doesn't raise an error if the commands don't quit properly, which they won't in some cases.

mpratt14, 2019-04-03:
@Ivoah do you have the exact error you receive?

I looked at the documentation:
https://docs.python.org/2/library/subprocess.html

There doesn't seem to be a large difference between call and check_call...
This is a command that is supposed to run once and be done...so these lines wouldn't cause an error in the middle of program running, right?

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada