adafruit/Adafruit_Python_GPIO #42: Can PCF8574 be used for LCD control with CharLCD library?

bgewehr, 2016-05-16:
The MCP23XXX GPIO extender is used as custom gpio for the CharLCD class.

Can the PCF8574 also be used as custom gpio for that purpose? (are all needed methods worked out?)
After my first attempts i cannot get it to work although code is executed without errors.

ladyada, 2016-05-16:
totally different chip, so you'd have to adapt the code for the different expander - check the datasheet and make a port! :)

closed 2016-05-16 by ladyada

bgewehr, 2016-05-17:
ladyada: there is a PCF8574 module in the gpio repository. in the lcd examples in your lcd repository the mcp230xx is used as a custom gpio to address the LCD. My question was if the module you present here for the PCF8574 is designed to be used as well as the mcp230xx to represent a custom gpio for the lcd modules. since this is a yes or no question, would you be so kind to give your opinion?

ladyada, 2016-05-17:
we didnt write that code, so we dont know.