adafruit/Adafruit_Python_GPIO #47: Would it be possible for support for the FT232H chip to be added?

mecworks, 2016-07-22:
Having support for a usb->GPIO breakout would be great for prototyping/rapid development on a workstation.

Support for something like the FT232H or a USB-RPi compatible breakout board like one of these:

http://garagetech.tips/rtk-gpio-usb-raspberry-pi-compatible-gpio-header/
http://garagetech.tips/rtk-gpio-usb-raspberry-pi-compatible-gpio-header/

In fact, it would be awesome if Adafruit would make something like one of those two boards.

If only support for the FT323H could be added, code could be run on from a PC workstation running Linux, a RPi or a BB.

mecworks, 2016-08-01:
Specifically, I'm talking about abstraction of the I2C layer over the FT232H so that the bus could be used interchangeably with the RPi, BB or FT232H without too much effort on the user's side.

tdicola, 2016-08-07:
Oh it's actually in there already--check out ft232h.py: https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/FT232H.py  I don't have a ton of examples of using it (check out the guide: https://learn.adafruit.com/adafruit-ft232h-with-spi-and-i2c-libraries) but I did intend for it to be a drop in replacement for the GPIO, I2C, and SPI interfaces that our other libraries use.  The FT232H class implements the GPIO base class functions for setting pins, digital IO, etc. The SPI class implements the interface for SPI devices that other libraries use, and the I2CDevice class implements the interface for I2C devices.  Check the guide above for more details, thanks!

closed 2016-08-07 by tdicola

mecworks, 2016-08-08:
Thanks, that helps a lot.  I'll dig into it.  On a side note, while installing the new python libraries onto a desktop system (so I could develop and test RPi code using the FT323H on a desktop system) I found that many of them do not install.  I entered the following issue:

https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/issues/156