adafruit/Adafruit_Python_GPIO #48: Adafruit_I2C.reverseByteOrder() in Adafruit_I2C.py is broken

pdp7, 2016-08-21:
@tdicola: this issue for `Adafruit_I2C.reverseByteOrder()` was raised by @phrogger:
https://github.com/adafruit/adafruit-beaglebone-io-python/issues/82
> The method Adafruit_I2C.reverseByteOrder() in Adafruit_I2C.py does not work. 
>
> It tries to automatically detect the presence of a 16 bit or 32 bit value by detecting the first byte containing non-zero bits. This fails whenever there are valid leading bytes containing zeros.

@jwcooper commented in [adafruit-beaglebone-io-python/Adafruit_I2C.py](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/Adafruit_I2C.py#L8):
> Adafruit_I2C.py is essentially a fork of the Adafruit Raspberry Pi I2C module. Any pull requests for this module should be directed to the following, and I can pull them. I'd rather not deviate from the original: https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/master/Adafruit_I2C

However, `Adafruit_I2C` now resides in this repo. `reverseByteOrder()` is located in [Adafruit_GPIO/I2C.py](https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/I2C.py#L28)

ahogen, 2017-08-07:
*crickets*

tdicola, 2017-08-23:
I'm going to close this one as won't fix as the reverseByteOrder function is now effectively dead code that isn't used by the updated Adafruit_Python_* libraries (the libraries from this README:  https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/).  The original bug https://github.com/adafruit/adafruit-beaglebone-io-python/issues/82 against the older Adafruit_I2C.py source for this function gets at the core issue that it's not really possible to make this function work for all 16-bit and 32-bit input.  During the refactoring that created the updated libraries they switched from using this reverseByteOrder function to doing explicit byte order changes with the Python struct module. With no more dependencies let's plan to delete this function entirely.  For now I'll change the function to throw a warning/error to catch if any dependency was missed and warn of the deprecation.

closed 2017-08-23 by tdicola

pdp7, 2017-08-24:
@tdicola thanks for handling this issue