adafruit/Adafruit_Python_GPIO #49: Beaglebone Black Jessie returns the wrong I2C bus value and cannot communicate on the default bus.

jwiki95, 2016-08-29:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  BEAGLEBONE BLACK DEBIAN JESSIE [KERNEL VERSION 4.4.16-ti-r38]

- Python version (run `python -version` or `python3 -version`):  PYTHON 2.7.9

- Error message you are receiving, including any Python exception traces: 
>>> import Adafruit_LED_Backpack.SevenSegment as ss
>>> seg = ss.SevenSegment(address=0x72)
>>> seg.begin()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_LED_Backpack/HT16K33.py", line 53, in begin
    self._device.writeList(HT16K33_SYSTEM_SETUP | HT16K33_OSCILLATOR, [])
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/I2C.py", line 127, in writeList
    self._bus.write_i2c_block_data(self._address, register, data)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_PureIO/smbus.py", line 274, in write_i2c_block_data
    self._device.write(data)
IOError: [Errno 110] Connection timed out

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
Try to address anything on I2C using existing module

In the newest version of the kernel for the BBB, the default address on pins P9_19 and P9_20 now addresses i2c-2 instead of i2c-1. This can be mitigated by checking which version of the kernel the is being run on the BBB and returning the appropriate value.

jwiki95, 2016-08-29:
```
root@beaglebone:~# ls /dev | grep i2c*
i2c-0
i2c-1
i2c-2
root@beaglebone:~# i2cdetect -r -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@beaglebone:~# i2cdetect -r -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- 72 -- -- -- -- --
root@beaglebone:~# uname -a
Linux beaglebone 4.4.16-ti-r38 #1 SMP Wed Aug 3 17:12:46 UTC 2016 armv7l GNU/Linux
root@beaglebone:~#

```

silver2row, 2018-03-04:
Hello...I do not know everything you are trying to do. 

But...in hindsight, I can work w/ you if necessary.  I have time and effort. I have been using the MotorBridgeCape from Seeed-Studio w/ a BBB/BBG/BBGW and so on. I have the Cape working at times and w/ specific boards. The current issue I have is w/ the Cape, which can be found at https://github.com/Seeed-Studio/MotorBridgeCapeforBBG_BBB, is that I cannot make it run my software as of now, e.g. as usual w/ my BBB and variants.

So,

About that i2c issue, sometimes adding a busnum (bus number) is helpful in specific cases.

I do not know your software nor do I have access to what you are doing while looking over this post.

Seth

P.S. If you need support or if you have given up, please shout back at me and we can either conclude or start. I hope to hear from you. Good day, sir.

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada