adafruit/Adafruit_Python_GPIO #50: TypeError:ftdi.write_data() function takes exactly 2 arguments (3 given)

lifetyper, 2016-10-10:
I met this error ever since I installed libftdi version 1.3 by homebrew,while evertying works fine in libftdi version 1.2.

By looking in to the libftdi python wrapper,the ftdi.write_data() changed from ：
`
def write_data(ftdi, buf, size):`
to:
`def write_data(context,buf)`
 
This change caused this error.

I tried to modify adafrui python gpio library to match the new libftdi,but failed in usb bulk write stage.

dennischrist, 2016-11-10:
Hi lifetyper, 

did you manage to solve this?

lifetyper, 2016-11-10:
well,currently the only way to make this working is roll back libftdi to version 1.2

dennischrist, 2016-11-10:
btw I'm looking for a script that can manage a 7pin 128x64 Oled, do you maybe know how to fix that?

tuescott, 2017-01-30:
I too am having this problem on OSX-Yosemite 10.10.5.  Although unlike lifetyper above, when I change Adafruit_GPIO.FT232H's call to ftdi.write_data()  to pass 2 args instead of 3, I get:
              RuntimeError: Timeout while polling ftdi_read_data for 2 bytes!
from mpsee_sync().  But since mpsee_sync()   (which wants to write itself a bad opcode and then look for the assoc. error as a way to sync itself) also uses the problematic write_data() function it is likely that the write never gets there now (that I'm yanking it around).   Or ??  

I guess my biggest question is: given this numberOfArguments mismatch between Adafruit_GPIO.FT232H  and the ftdi.write_data() in libftdi, how does this code work for anyone???

tdicola, 2017-02-07:
Ah that's annoying, it appears libftdi made a breaking change to the API in their latest 1.3 version.  Rolling back to the 1.2 version is the best workaround for now, with homebrew you should be able to run these commands to uninstall libftdi 1.3 and reinstall from an older 1.2 version:
```
brew uninstall libftdi
brew install --build-from-source https://raw.githubusercontent.com/Homebrew/homebrew-core/99ebd41decde2dd0fbab799bed88cfcb230a096a/Formula/libftdi.rb
```
Give that a shot, it should give you the 1.2 version again that's made to work with the current code.  I'll have to look later at how to support both versions in the code--it could be tricky since that's a pretty big breaking change they seem to have made to the python API. :/

tuescott, 2017-02-07:
Thanks tdicola, for the sanity check.
So I did as you suggested, but now I can no longer import ftdi1....

Here is my traceback on a 10.10.5 Yosemite machine: doing a simple interactive import of ftdi1:
python
Python 2.7.13 (v2.7.13:a06454b1afa1, Dec 17 2016, 12:39:47) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ftdi0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ftdi0
>>> import ftdi1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/ftdi1.py", line 20, in <module>
    _ftdi1 = swig_import_helper()
  File "/usr/local/lib/python2.7/site-packages/ftdi1.py", line 19, in swig_import_helper
    return importlib.import_module('_ftdi1')
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _ftdi1

gorzilla, 2017-09-20:
Is this an upstream bug? I can't see any reason for the change.

SolidTux, 2017-10-19:
It's especially strange as there is no mention in the [changelog](https://www.intra2net.com/en/developer/libftdi/index.php).

GHPS, 2018-02-06:
> It's especially strange as there is no mention in the changelog.

Yes - the function signature of write_data has changed between version
1.2 and 1.3 due to the fix of a Python 3 swig bug. Apparently unnoticed by the
developers since even their sample code complete.py has this bug, see lines 63, 66, 71.

rurigeo, 2018-07-05:
try liftdi1-1.4 , works for me just remove length form function write_data()

peci1, 2019-02-07:
I submitted a bug to Ubuntu packaging to provide libftdi 1.2 for python-ftdi and libftdi 1.4 for python3-ftdi.

https://bugs.launchpad.net/ubuntu/+source/libftdi1/+bug/1815049

mpratt14, 2019-03-26:
Fixed with #93 ...

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada