adafruit/Adafruit_Python_GPIO #54: nvidia Jetson TX1 support

shoemakerlevy9, 2016-12-07:
Enhancement: 

I am running this library on a TX1 along with the Adafruit_Python_ADS1x15 library in order to use the ADS1015 chip through the built in I2C port. 

I modified the I2C.py file, replaced the raise on line 55 with a return 0 and it works, which suggests that the default I2C bus on the TX1 board is 0 (like the original raspberry pi). 

If support for the TX1 where added to the platform_detect() function then the I2C bus could be dynamically set here.

tdicola, 2018-02-27:
Oh interesting, that's cool you're trying out the Jetson board!  I'm going to close this though as it's not really tracking a defect or problem with the library.  Thanks!

closed 2018-02-27 by tdicola

sharoseali, 2018-10-28:
@shoemakerlevy9  can u completely explain how u modify this code for TX1??? i want to use it with TX2 i need Your help....