adafruit/Adafruit_Python_GPIO #61: Not work for python3 with too many problems...

dukelec, 2017-04-03:
First:

```
Python 3.5.2 (default, Jul 28 2016, 18:18:57) 
[GCC 5.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> import Adafruit_GPIO as GPIO
>>> import Adafruit_GPIO.FT232H as FT232H
>>> FT232H.use_FT232H()
>>> ft232h = FT232H.FT232H()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 165, in __init__
    self._mpsse_sync()
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 249, in _mpsse_sync
    data = self._poll_read(2)
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 229, in _poll_read
    raise RuntimeError('Timeout while polling ftdi_read_data for {0} bytes!'.format(expected))
RuntimeError: Timeout while polling ftdi_read_data for 2 bytes!
>>> 
```

after comment out this line: `self._mpsse_sync()`:

```
Python 3.5.2 (default, Jul 28 2016, 18:18:57) 
[GCC 5.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_GPIO as GPIO
>>> import Adafruit_GPIO.FT232H as FT232H
>>> FT232H.use_FT232H()
>>> ft232h = FT232H.FT232H()
>>> 
>>> ft232h.setup(8, GPIO.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 341, in setup
    self.mpsse_write_gpio()
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 313, in mpsse_write_gpio
    self._write(self.mpsse_gpio())
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 309, in mpsse_gpio
    return str(bytearray((0x80, level_low, dir_low, 0x82, level_high, dir_high)))
TypeError: an integer is required
>>> 
```

after remove all `chr` in function `mpsse_gpio`:

```
Python 3.5.2 (default, Jul 28 2016, 18:18:57) 
[GCC 5.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_GPIO as GPIO
>>> import Adafruit_GPIO.FT232H as FT232H
>>> FT232H.use_FT232H()
>>> ft232h = FT232H.FT232H()
>>> 
>>> ft232h.setup(8, GPIO.OUT)
>>> ft232h.output(8, GPIO.HIGH)
>>> ft232h.output(8, GPIO.LOW)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 368, in output
    self.mpsse_write_gpio()
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 313, in mpsse_write_gpio
    self._write(self.mpsse_gpio())
  File "/mnt/TT/SA/tools/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py", line 194, in _write
    raise RuntimeError('ftdi_write_data failed with error {0}: {1}'.format(ret, ftdi.get_error_string(self._ctx)))
RuntimeError: ftdi_write_data failed with error -1: usb bulk write failed
>>> 
```

Adafruit_Python_GPIO version:
```
commit 060f6d6782768b1f6a4713f369db2740e79ec61e
Author: Tony DiCola <tony@tonydicola.com>
Date:   Mon Mar 6 22:44:49 2017 -0800

    Minor bump for Pi 3 on 4.9.x kernel detection.
```

libftdi vertion:
`
libftdi1-1.2
`

python version:
`
Python 3.6.0
Python 3.5.2
`

no problem with `Python 2.7.12`

atelier-ritz, 2017-09-07:
I also had the same problem on Python 3.6, so I did some minor fix to "FT232h.py". Although I only tested GPIO and I2c features, it is working fine so far. You can replace FT232h.py in your Adafruit_GPIO folder by this one https://github.com/atelier-ritz/Python3.6-32bit--FT232H-libftdi1.4. If you are interested in what changes are made, you can google "Python3 bytes bytearray unicode" and it will give you a rough idea.

GHPS, 2018-02-06:
Short answer: The problem seem be caused by libftdi version 1.2 and Python 3.x. 
Updating to version 1.4 (or at least to 1.3) and upgrading FT232H.py solves the problem.
So you need a new version of libftdi with Python 3.x or you stay with version 1.2 and Python 2.x -
but don't mix them.

Since a lot of the Adafruit_GPIO users are bitten by this bug it's worthwhile to take
a look at the full picture

Long version:

Under Kubuntu 16.04-64 I had the same problem. I started with the current version of the Adafruit_GPIO library (commit 22a1ff4 version), Python 3 (version 3.5.2) and with libftdi1-2 (1.2-5build1) which is available from the official Ubuntu ressources.
This resulted is the known timeout error when instantiating the board:

```
<ipython-input-4-b5c894ce1f4c> in <module>()
----> 1 ft232h = FT232H.FT232H()

/Download/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py in __init__(self, vid, pid, serial)
    163         # Enable MPSSE and syncronize communication with device.
    164         self._mpsse_enable()
--> 165         self._mpsse_sync()
    166         # Initialize all GPIO as inputs.
    167         self._write('\x80\x00\x00\x82\x00\x00')

/Download/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py in _mpsse_sync(self, max_retries)
    247         sync = False
    248         while not sync:
--> 249             data = self._poll_read(2)
    250             if data == '\xFA\xAB':
    251                 sync = True

/Download/Adafruit_Python_GPIO/Adafruit_GPIO/FT232H.py in _poll_read(self, expected, timeout_s)
    227                 return str(response)
    228             time.sleep(0.01)
--> 229         raise RuntimeError('Timeout while polling ftdi_read_data for {0} bytes!'.format(expected))
    230 
    231     def _mpsse_enable(self):

RuntimeError: Timeout while polling ftdi_read_data for 2 bytes!
```

1) Missing Python 3 support of Adafruit_GPIO lib?

In the inital post dukelec had the same error message so the conclusion lies at hand that
the missing Python 3 support of Adafruit_GPIO lib could be the cause of the problem.

Therefore I swapped the original FT232H.py with atelier-ritz's version (thanks for providing the patches!).
But the new FT232H.py relies on libftdi 1.4 and I still had 1.2 installed. Between 1.2 and 1.3 the
function signature of write_data changed [1] - a problem also noticed by the people who tried newer
versions of libftdi with Adafruit_GPIO lib [2] (and - even worse - can be found in libftdi's 
own example code complete.py, lines 63, 66, 71). So I had to make some changes to the new FT232H.py 
starting with line 188:

`ret = ftdi.write_data(self._ctx, string)`

to

`ret = ftdi.write_data(self._ctx, string, length)`

After some minor changes elsewhere the code ran fine under Python 3 - and produced exactly the same
time-out error as before. So that was a trip in a roundabout - with the hunch that the Adafruit_GPIO 
lib is not the cause of the problem. 

2) Python 3 support in libftdi?

So I rolled back the changes I made myself and returned to the new, original FT232H.py and started experimenting  with various versions of libftdi. As suggested here[3] I installed libftdi from source (after uninstalling the official Ubuntu version). And, beginning with libftdi version 1.3 the time out error was gone. According to the change log this was probably due to a swig bug in libftdi.
Here is the updated version of Adafruit's install instructions (for Python 3.5 and libftdi 1.4).

```
sudo apt-get update
sudo apt-get install build-essential libusb-1.0-0-dev swig cmake python-dev libconfuse-dev libboost-all-dev
wget http://www.intra2net.com/en/developer/libftdi/download/libftdi1-1.4.tar.bz2
tar xvf libftdi1-1.4.tar.bz2
cd libftdi1-1.4
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX="/usr/" -DPYTHON_INCLUDE_DIR="/usr/include/python3.5" -DPYTHON_LIBRARIES="/usr/lib/python3.5/" ../
make
sudo make install
```

So that's it...

GHPS

P.S. @atelier-ritz Please submit an official pull request - even though Adafruit is years behind 
merging the code that other people already fixed.

1: http://developer.intra2net.com/git/?p=libftdi;a=commitdiff;h=f64b66d604c5d78886d47034fb2db2176fc24a09
2: https://github.com/adafruit/Adafruit_Python_GPIO/issues/50
3: https://learn.adafruit.com/adafruit-ft232h-breakout/linux-setup

atelier-ritz, 2018-02-06:
Thank you GHPS for your detailed report.
I will submit an official pull request when I have time :D

shapecoder, 2019-05-24:
Is anybody working on this issue? We have integrated the FT232H board on some of our electronics but the laptop controlling the thing crashed. We figured we would try to get it running on Python 3 64-bit, since 2 is so old (we're not even able to find PyQt5 Wheels for it anymore). Turns out this is much more difficult than expected.

Switching to the updated libftdi seems to work on Python 3.6 by using https://sourceforge.net/projects/picusb/files/libftdi1-1.4git_devkit_x86_x64_14June2018.zip/download

However, Adafruit_GPIO has a few issues:
- Tabs/spaces errors, relatively easy to fix
- string errors where bytes are expected, also fixable with .encode()
- timeout errors as described by people in this issue

We haven't tried the atelier-ritz update yet (https://github.com/atelier-ritz/Python3.6-32bit--FT232H-libftdi1.4) or the "make FT232H compatible with Python 3" branch (https://github.com/adafruit/Adafruit_Python_GPIO/pull/106). Is there anybody who has experience with these solutions and/or whether they will be merged in the official release soon?

Barring finding PyQt5 for Python 2 we'd have to run the FT232H code in Python 2 and the rest in Python 3 or something. It would be nice to be able to just use the FT232H with Python 3 64-bit...

atelier-ritz, 2019-05-24:
Hi, I don't know if Adafruit is currently working on this issue, but I had
the same issues as you do when I started using FT232H board on Python 3 on
a Windows machine. Basically I dealt with the string/byte errors (as you
mentioned, using .encode()) in FT232H.py and did some workaround on libftdi
version... Finally I could get FT232H to work with Python3 and PyQt5. (It
took me about a week though) So it's definitely possible, but I agree with
you that Adafruit should post a version that works with Python 3.6+.

I have FT232H working on Windows but I'm not sure about Ubuntu cuz I didn't
try it.

atelier-ritz

ladyada, 2019-05-24:
this repo def need some attention. if anyone can help by submitting a PR to make it py3 compatible and also testing, it would be deeply appreciated!

shapecoder, 2019-05-24:
I just tested the existing PR (https://github.com/adafruit/Adafruit_Python_GPIO/pull/106) and it seems to work :)

Installed Python 3.6.8 and used the libftdi1-1.4 driver (installed everything according to the manual). Pulled the Adafruit_GPIO master and replaced the FT232H.py from the PR before installing and I'm able to ping my ADS1115 ADC and read out a channel. Really happy that it works, saved my weekend :P

ladyada, 2019-05-24:
please comment on that saying what you ended up testing - we haven't merged it cause there's not a lot of test coverage (yet)

mpratt14, 2019-05-25:
@ladyada I would recommend merging #106 

The author explained the changes really well in the conversation I had with him, and I did reading on python 3 documentation to verify. They are really small changes really, but necessary for python 3, and equivalent for python 2

ladyada, 2019-05-26:
done, we'll see who complains it doesnt work - will revert if necessary

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada