adafruit/Adafruit_Python_GPIO #62: Getting SPI error after Kernel module reload

fiveseven808, 2017-05-01:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): 
 Raspberry Pi with Raspbian operating system Kernel 4.9.24+ #991

- Python version (run `python -version` or `python3 -version`): 
Python 2.7.9

- Error message you are receiving, including any Python exception traces: 
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/SPI.py", line 42, in __init__
IOError: [Errno 2] No such file or directory

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

1.  Instantiate display via the Adafruit SSD1306 library. 
2. Load FBTFT kernel modules
3. Remove FBTFT kernel modules 
4. Reload the spi_bcm2835 kernel module 
5. Try to reinstantiate display via Adafruit SSD1306 
6. Get error message 

Wondering if this is a Kernel 4.9 thing? It was working with Kernel 4.4

Johnrasp, 2018-02-20:
Exactly the same error using Ubuntu 16.4 on Raspberry Pi 3
File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/SPI.py", line 42, in __init__
    self._device.open(port, device)
IOError: [Errno 2] No such file or directory
No found a solution to this yet. So SPI GPIO use impossible

tdicola, 2018-02-27:
If you're hitting a file not found or IOError no such file then it means the /dev/spidev devices aren't available.  If you're starting from scratch and have made no modifications to the Pi, be sure you've run raspi-config and enabled the SPI bus (as mentioned in the guides for using different add-ons).  If you're not using Raspbian and instead using Ubuntu I'm not sure what the right process to follow is, but see if there's an analog to raspi-config or try using Raspbian.  

If you're making modifications and loading and unloading kernel modules that could definitely remove the SPI devices too.  These days I believe the Pi kernel prefers to use device tree overlays to enable and disable SPI ports so it's not clear to me if reloading the kernel module is enough to turn them back on after being disabled--you might need to dig into device tree overlays and updates.  I'm going to close this issue though as it's not something with the library itself--in general as guides mention you need to make sure SPI support is enabled on the device, typically with raspi-config.

closed 2018-02-27 by tdicola