adafruit/Adafruit_Python_GPIO #65: Incorrect detection of Raspberry Pi version

ameer1234567890, 2017-05-10:
Platform.py detects Raspberry Pi version on Kernel 4.9 incorrectly. It assumes BCM2708 as Pi 1, BCM2709 as Pi 2 and BCM2835 as Pi 3. In contrast BCM2835 is Pi 1, BCM2836 is Pi 2 and BCM2837 is Pi 3 on kernel 4.9.

arcSlayer85, 2017-06-01:
could this be why my lcd code thinks my RPi 2B needs a BeagleBone GPIO module? when it doesn't

slokhorst, 2017-06-06:
My Pi 2 is listed as `BCM2835` in kernel 4.9.

ladyada, 2017-06-06:
btw Pi 2's now come with a Pi 3 chipset so they're essentially a Pi 3 minus BT/WiFi

tdicola, 2017-06-06:
@arcSlayer85 Make sure to update the library to the latest version, #60 fixed the kernel detection issue.  Check the bug for details on how to force an upgrade in case you've got an older version of the library and python doesn't want to update it.

Also the original issue from @ameer1234567890 should be resolved with #60 too.  For anyone running into an error make sure to download and reinstall the GPIO library with the force option like #60 mentions.

closed 2017-06-06 by tdicola