adafruit/Adafruit_Python_GPIO #73: error accessing p8_7 and p8_8 on BBB

mjkkirschner, 2017-09-11:
- Platform/operating system:  **Beagle Bone Black - Angstrom image**

- Python version:  **2.7**

- Error message you are receiving, including any Python exception traces:  **~~GPIO set failed - insufficient permissions or file does not exist.***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
 * install old angstrom linux image, 
* install python GPIO using tutorial on adafruit for angstrom linux,  
* write any python script that tries to set pins 7 or 8 as GPIO pins `GPIO.Setup("P8_7",GPIO.OUTPUT)
* error above occurs on run using python
* I am logged in as root user
* pins P8,10-18 seem to work.

mjkkirschner, 2017-09-11:
after staring at some documentation for the headers it seems I should probably be using:
`P8_07` - will test and close if thats the case.

nickamuch, 2017-11-06:
The issue has to do with the fact that GPIO 66 and 67 are used for a green and red led.   The underlying c file tries to access a file relating to the brightness of these LED's, which may not exist.  I modified the following #defines in event_gpio.h, setting them to 51 and 52 or something that won't be accessed. 

#define USR_LED_RED      66
#define USR_LED_GREEN    67

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada