adafruit/Adafruit_Python_GPIO #80: Adafruit_Python_GPIO not compatible with Python 3.6.1, only Python 2.7.9

Ragooman, 2018-02-18:
I have both Python 2.7.9 and Python 3.6.1 installed on my RPi2. I'm using the MPC3008 interfaced to the hardware port for SPI. And I use the example program from Adafruit_Python_MPC3008 library called "simpletest.py".
This works perfectly when I use Python 2.7.9. 

However, when I try using Python 3.6.1, I get this error message
**ModuleNotFoundError: No Module named 'Adafruit_GPIO'**
As I mentioned, I'm using the hardware port for the SPI. So I followed the instructions to uncomment those steps and comment out the steps for the software port.

I installed the MPC3008 library using the instructions from this homepage on github
https://github.com/adafruit/Adafruit_Python_MCP3008 
And I installed the GPIO library using the instructions from your homepage here on github.
https://github.com/adafruit/Adafruit_Python_GPIO

Unless I'm mistaken, I couldn't find any solutions this problem in the previous issues.
I'm trying to merge this example code into my new project, but I'm using Python3 with networking code. And I like to remain with this version. 

below is the code showing what I did

```
# Simple example of reading the MCP3008 analog input channels and printing
# them all out.
# Author: Tony DiCola
# License: Public Domain
import time

# Import SPI library (for hardware SPI) and MCP3008 library.
import Adafruit_GPIO.SPI as SPI
import Adafruit_MCP3008

# Software SPI configuration:
#CLK  = 18
#MISO = 23
#MOSI = 24
#CS   = 25
#mcp = Adafruit_MCP3008.MCP3008(clk=CLK, cs=CS, miso=MISO, mosi=MOSI)

# Hardware SPI configuration:
SPI_PORT   = 0
SPI_DEVICE = 0
mcp = Adafruit_MCP3008.MCP3008(spi=SPI.SpiDev(SPI_PORT, SPI_DEVICE))

print('Reading MCP3008 values, press Ctrl-C to quit...')
# Print nice channel column headers.
print('| {0:>4} | {1:>4} | {2:>4} | {3:>4} | {4:>4} | {5:>4} | {6:>4} | {7:>4} |'.format(*range(8)))
print('-' * 57)
# Main program loop.
while True:
    # Read all the ADC channel values in a list.
    values = [0]*8
    for i in range(8):
        # The read_adc function will get the value of the specified channel (0-7).
        values[i] = mcp.read_adc(i)
    # Print the ADC values.
    print('| {0:>4} | {1:>4} | {2:>4} | {3:>4} | {4:>4} | {5:>4} | {6:>4} | {7:>4} |'.format(*values))
    # Pause for half a second.
    time.sleep(0.5)
```

Ragooman, 2018-02-26:
I was able to resolve this problem with help from the admin on the Adafruit forum,
here's the link to the results
https://forums.adafruit.com/viewtopic.php?f=50&t=131286&p=653870#p653870

closed 2018-02-26 by Ragooman

tdicola, 2018-02-26:
Yep make sure to install into the right python version--it can be tricky sometimes with both python 2 and 3 installed.  If in doubt always run pip3 to install for python3 and pip2 to install for python2 (or just `python` if you aren't sure which version you have, it's probably 2.x right now).