adafruit/Adafruit_Python_GPIO #81: no module named 'Adafruit_GPIO' (even though I checked everything)

GawiQ, 2018-03-02:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspbian GNU/Linux 9 (stretch)

- Python version (run `python -version` or `python3 -version`):  python 2.7.13 and 3.5.3

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "/home/pi/Desktop/python-lis3dh-master/LIS3DH_bus1.py", line 2, in <module>
    from Adafruit_GPIO import I2C
ImportError: No module named 'Adafruit_GPIO'

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
- installed Adafruit_GPIO for both python and python3 (it appears on the lists of pip and pip3). 
- checked for RPi.GPIO and it is installed as well
- still after just first line which is 'from Adafruit_GPIO import' I get the error I mentioned, no idea what's wrong
- checked every step, power is adequate, and 'sudo i2cdetect -y 1' detects the device

chrisb2, 2018-03-02:
Thanks for the detail, can you provide a simplified version of LIS3DH_bus1.py which replicates this issue and I will give it a go myself. 

So far the symptoms do not look like my library is actually the root cause of the issue, more like something wrong with the package installation on your system. It might be a idea to also raise this issue in the appropriate Raspberry Pi forum. If you do this send me a link.

regards,
Chris

closed 2018-03-04 by GawiQ

GawiQ, 2018-03-11:
This turned out to be some system error. Reinstalling the system was the solution.