adafruit/Adafruit_Python_GPIO #87: FTH232H.py has import ftdi1 as ftdi

uchuugaka, 2018-08-09:
There seems to be no such library as ftdi1 that FTH232H.py is trying to import.
pip doesn't seem to know about it.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): 
macOS

- Python version (run `python -version` or `python3 -version`):  
2.7.10

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "./ft232h.py", line 4, in <module>
    from Adafruit_GPIO import (
  File "/Users/myusername/Library/Python/2.7/lib/python/site-packages/Adafruit_GPIO/FT232H.py", line 30, in <module>
    import ftdi1 as ftdi
ImportError: No module named ftdi1

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
Try to import like
from Adafruit_GPIO import FT232H

applefreak, 2018-11-20:
Any luck on this issue? I'm getting the same thing

jasperSpeicher, 2018-12-05:
I also get this issue... High Sierra, Python 3.7.0, pyenv

uchuugaka, 2018-12-05:
I literally just gave up a few months ago on this.
I have two of these units.
They are exactly what I needed and wanted to use, but I could not figure out how to get them to actually work and the steps seemed to make things worse on the mac itself.
It would be great if they could update this stuff.
Maybe then they could sell more of it too.

ladyada, 2018-12-05:
this library is about to be deprecated, MacOS / FTDI keeps breaking stuff :(

uchuugaka, 2018-12-05:
Aww. Sad to hear that.
I'm not terribly surprised.
Is there any alternative that might make it work on the C/C++ side, or a recommendation we could work on to contribute to keeping it working?
I'd be happy to put in the time if I could be pointed in the right direction.
It definitely sounds like the C/C++ underpinnings need to be bundled with the Python library.

peci1, 2019-02-05:
Guys, you can compile the library yourself (commands for Ubuntu):

    sudo apt-get update
    sudo apt-get install build-essential libusb-1.0-0-dev swig cmake python-dev libconfuse-dev libboost-all-dev
    wget http://www.intra2net.com/en/developer/libftdi/download/libftdi1-1.2.tar.bz2
    tar xvf libftdi1-1.2.tar.bz2
    cd libftdi1-1.2
    mkdir build
    cd build
    cmake -DCMAKE_INSTALL_PREFIX="/usr/" ../
    make
    sudo make install

mpratt14, 2019-04-04:
@uchuugaka @peci1 @applefreak @jasperSpeicher 

Just wondering....have any of you followed the directions here to merge libftdi into your python env?
https://learn.adafruit.com/adafruit-ft232h-breakout/mac-osx-setup

Note: if you have el capitan, yosemite, mavericks you want to use the code in the directions
But if you have a newer system like mojave, sierra, high sierra, you want to use the newest master...
by substituting the following line with the one in the directions...

`brew install --build-from-source https://raw.githubusercontent.com/Homebrew/homebrew-core/master/Formula/libftdi.rb`

mpratt14, 2019-04-04:
@jasperSpeicher BTW it is not recommended to run this library on Python 3

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada