adafruit/Adafruit_Python_GPIO #91: Unable to read user registry using readRaw8 with Adafruit_GPIO.FT232H i2c connection

oshadmon, 2018-11-29:
**Machine Specs**: Ubuntu 18.04 Desktop on an i7 based workstation.

**Background**: I'm using an Adafruit HTU21D-F Humidity/Temp sensor connected to a USB pluggable FT232H breakout board (C232HM-DDHSL‌-0 USB2.0/I2C cable). Adafruit GPIO has support for FT232H (https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/FT232H.py)

**Question**: I found a [smbus script](https://github.com/bursam/htu21d-f/blob/master/HTU21DF.py) to retrieve data from my sensor; however, would much prefer to use the _Adafruit_GPIO.FT232H_ module. My issue is when trying to read user registry I get an error saying `RuntimeError: Failed to find expected I2C ACK!`. Anybody know why that is? How would I fix it? I have already added a pull-up resistor.  

**Code** 
```import time
from Adafruit_GPIO import FT232H

HTU21DF_ADDRESS        = 0x40
HTU21DF_READTEMPCMD    = 0xE3
HTU21DF_READHUMCMD     = 0xE5
HTU21DF_READUSERREGCMD = 0xE7
HTU21DF_SOFTRESETCMD   = 0xFE

# Temporarily disable FTDI serial drivers.
FT232H.use_FT232H()
# Find the first FT232H device
ft232h=FT232H.FT232H()
# Create an I2C device at address 0x70.
i2c = FT232H.I2CDevice(ft232h, HTU21DF_ADDRESS)
print(i2c.ping())

# Soft Reset 
i2c.writeRaw8(HTU21DF_SOFTRESETCMD)
time.sleep(0.015)

# read_user_reg
i2c.writeRaw8(HTU21DF_READUSERREGCMD)
print(i2c.readRaw8()) # Error occurs on this line

```
**Execution & Full error**
```
ubuntu@sonoma:~$ sudo python3 setup_adafruit_FT232H.py 
[sudo] password for ubuntu: 
True
Traceback (most recent call last):
  File "setup_adafruit_FT232H.py", line 27, in <module>
    print(i2c.readRaw8())
  File "/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/FT232H.py", line 743, in readRaw8
    self._verify_acks(response[:-1])
  File "/usr/local/lib/python3.6/dist-packages/Adafruit_GPIO-1.0.3-py3.6.egg/Adafruit_GPIO/FT232H.py", line 647, in _verify_acks
    raise RuntimeError('Failed to find expected I2C ACK!')
RuntimeError: Failed to find expected I2C ACK!```

zrnsm, 2019-03-28:
I just wasted a bunch of time on this same exact bug. Here's a possible workaround that seems to work for me reading from a simple custom STM32 slave: 
```python
    def readRaw8(self):
        """Read an 8-bit value on the bus (without register)."""
        self._idle()
        self._transaction_start()
        self._i2c_start()
        # Possible workaround
        #self._i2c_write_bytes([self._address_byte(False)])
        #self._i2c_stop()
        #self._i2c_idle()
        #self._i2c_start()
        self._i2c_write_bytes([self._address_byte(True)])
        self._i2c_read_bytes(1)
        self._i2c_stop()
        response = self._transaction_end()
        self._verify_acks(response[:-1])
        return response[-1]
```
The extra write-stop-idle-start cycle without the r/w bit set was the culprit.

mpratt14, 2019-03-29:
#67 looks very related, might want to read it

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada