adafruit/Adafruit_Python_GPIO #99: NameError: name 'ensurepip' is not defined

wolfer649, 2019-04-01:
"NameError: name 'ensurepip' is not defined" - msg received when running setup.py.
Platform: Raspberry Pi 3 B with fresh Rasbian Stretch 06-27 install.
Adafruit_Python_GPIO was cloned from github.
Line 1 of setup.py is "ensurepip" which fails with the error message above.
If line 1 is commented out, the script seems to run OK for Python 3 and Python 2.7.
(the "python" command on this version of Stretch defaults to Python 2.7)
BTW, this was the root cause of an error installing the Adafruit MAX31855 software.

ladyada, 2019-04-02:
@mpratt14 any suggestions?

ladyada, 2019-04-02:
@wolfer649 meanwhile please use https://learn.adafruit.com/thermocouple/python-circuitpython

RyAndrew, 2019-04-02:
This is really bumming me out.
It seems this commit added "ensurepip"
https://github.com/adafruit/Adafruit_Python_GPIO/commit/57adb1e500de4649c16eb2743e9f2d47aeb64d1d#diff-2eeaed663bd0d25b7e608891384b7298

However ensurepip seems to be disabled:
pi@raspberrypi:~ $ python3 -m ensurepip --upgrade
**ensurepip is disabled in Debian/Ubuntu for the system python.**

RyAndrew, 2019-04-02:
Temporary workaround - install Adafruit_Python_GPIO from the commit before it was broken:
`pip install git+git://github.com/adafruit/Adafruit_Python_GPIO@0ec1977df23c0a043f6a53c12aea90ed79782b5d`

mpratt14, 2019-04-02:
3 things:

Yeah...my bad...I messed up the pull request...
In order for ensurepip to run in a script it has to be imported and called with the bootstrap() method...
I've been testing a lot today and I'm going to pull request to fix my mistakes...

There are some small issues left in the SPI methods that I worked on (there was more before that), also going to push that...this time I made sure I tested everything extensively.... (but I'm not testing on python 3 right now)
So you might not want to use the older version...but it's not hard to remove the .egg and reinstall

you are using python 3 it seems (which, I'm not sure will run this library well or not) so you don't need to worry about ensurepip. I'll have the positive existence of pip handled when I fix the setup.py

mpratt14, 2019-04-02:
Again, very sorry

See my requests #100 #101 

should be perfect now

ladyada, 2019-04-02:
they've been merged, please try!

RyAndrew, 2019-04-02:
thanks seems to be working

wolfer649, 2019-04-02:
It works great for Python 2.7 on Rasbian Stretch Full 06-27 w/Pi 3 B.

For Python 3, the only problem is that the print statements in setup.py are not Python 3 compatible.

i.e. no parens.

When I comment them out, the install runs fine with Python 3 and my MAX31855-based software works under Python3 as well.

Well, that's just my platform.

Thanks for the late night fix.

ladyada, 2019-04-02:
this library was never py3 compat - please use please use https://learn.adafruit.com/thermocouple/python-circuitpython :)

mpratt14, 2019-04-02:
@ladyada I think you keep linking the wrong thing...
do you mean this?
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/overview

@wolfer649 No problem, I'm just glad I was able to fix my own mistakes.

I had no idea the incompatibility between the python versions were so extreme! 
setup.py should maybe stay that way though...since this is a python 2 library only...

Maybe add that detail to the readme, and then a fully compatible print line to remind the user they should be using python 2.x in setup.py...

ladyada, 2019-04-02:
im pretty sure im linking to the right thing, its our Blinka-based MAX31855 library, if they follow those steps they'll be good to go :)

mpratt14, 2019-04-02:
@ladyada Ah...I looked over the part where @wolfer649 mentioned using the MAX31855.

@wolfer649 which library exactly did you try to install other than this one? or are you using something completely different to run that device? wrote your own?
There's so many of them that look similar....

Is it simply not compatible with python 3 or only the setup.py for that one?

I might have misunderstood, but does this library run well on python 3, at least the sections used for the thermocouple?

I'm guessing you are using this one, which seems to depend on this library
https://github.com/adafruit/Adafruit_Python_MAX31855

If anyone is trying to use CircuitPython for SPI or I2C in general uses / other devices I would start here:
https://learn.adafruit.com/circuitpython-basics-i2c-and-spi/spi-devices
https://circuitpython.readthedocs.io/en/3.x/shared-bindings/busio/SPI.html

wolfer649, 2019-04-02:
Apologies if I've caused a schmozzle here, re: Python 3.

Python 3 is no biggie for me. I was just trying to future-proof my code for Python 3, since Raspbian Stretch seems headed that way

and the default matplotlib install is now Python 3 as well.

(My project doc is in https://github.com/wolfer649/WGOT)

 

Just to summarize, in case it answers some questions …

I had installed the Adafruit MAX31855 software based on the instructions at the (now, no longer supported) web site …

https://learn.adafruit.com/max31855-thermocouple-python-library/software

Until recently, that had worked OK - for Python 2.7 and Python 3 - then failed recently, as described earlier.

And, as mentioned before, the MAX31855 install problem was caused by the 'ensurepip' statement in the setup.py for Adafruit_Python_GPIO.

As of now the MAX31855 software installation works great for Python 2.7, but with the "print" statement errors for Python 3 in setup.py for Adafruit_Python_GPIO.

My MAX31855-base code works fine with Python 3 now too, though I have to clone Adafruit_Python_GPIO and comment out the "print" statements to get that fully installed.

 

I will try the CircuitPython approach in the future, assuming it allows me to re-wire the MAX31855 connections to be compatible with the Adafruit 2.8" Capacitive Touch TFT - another topic for another time.

Thanks to all for your help and advice.

 

https://github.com/adafruit/Adafruit_Python_MAX31855

https://learn.adafruit.com/circuitpython-basics-i2c-and-spi/spi-devices
https://circuitpython.readthedocs.io/en/3.x/shared-bindings/busio/SPI.html

mpratt14, 2019-04-03:
@wolfer649 Thanks for the full explanation, there's so many things involved I got confused. Looks like it would indeed help you to make the print statements compatible with both so I'll do that right now

mpratt14, 2019-04-03:
@ladyada hopefully we are done with setup.py now
the print statements can be removed too, your choice, but this makes them run on both
#102

ladyada, 2019-04-03:
@mpratt14 awesome, thank you thank you!!!!

ladyada, 2019-09-30:
hiya thank you for the issue. we're going to be deprecating this library in favor of [our python3 Blinka library](https://github.com/adafruit/Adafruit_Blinka) which has [support for almost 200 different drivers](https://circuitpython.readthedocs.io/projects/bundle/en/latest/drivers.html), and a [wide variety of linux computers](https://circuitpython.org/blinka) as well as [ft232h
](https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h)
If this issue is still occuring when you use Blinka, please re-open it there!

closed 2019-09-30 by ladyada